ttu/ruuvitag-sensor #246: Deprecation warning in examples

terop, 2024-03-11:
Examples (and some code too) use the  `asyncio.get_event_loop().run_until_complete()` function call chain to run asynchronous functions. 
With Python 3.12 a DeprecationWarning is emitted: `ruuvitag-sensor/examples/get_async_bleak.py:18: DeprecationWarning: There is no current event loop`. Python [`get_event_loop()` documentation](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.get_event_loop) recommends to either use `get_running_loop()` or `asyncio.run()`. Using `get_running_loop()` results in the below error so it is not a straightforward replacement.
```
$ python3 examples/get_async_bleak.py
Traceback (most recent call last):
  File "<redacted>/ruuvitag-sensor/examples/get_async_bleak.py", line 18, in <module>
    asyncio.get_running_loop().run_until_complete(main())
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: no running event loop
```
I am wondering what would be the best way to fix the DeprecationWarning. 
There is no rush to fix this but fixing this before the call results in an error would be advisable.

**Environment:**
 - OS: Gentoo
 - ruuvitag_sensor package version: master (2.3.1)
 - RuuviTag firmware version: not relevant
 - Data format: not relevant

ttu, 2024-10-15:
Like you mentioned we should replace asyncio's `get_event_loop().run_until_complete()` with `run()`.

In RuuviTag's codebase, asyncio is not used in many places (main, Bleak, RX, examples, and documentation), but we should still ensure that everything works correctly after the change.

At least these come to my mind now. Can you think something I might have missed?

* Bleak works normally after the change to `asyncio.run()`.
* `RuuviTagReactice` uses `loop = asyncio.get_event_loop()`. Should we use `get_running_loop` here? [code](https://github.com/ttu/ruuvitag-sensor/blob/eb097f19ffca38b478f7297f01b68c853c66cb9f/ruuvitag_sensor/ruuvi_rx.py#L90) [example](https://github.com/ttu/ruuvitag-sensor/blob/db55ff1b5d9426dc44e5fcf84b3e2c7c1d08f84a/examples/rx_async.py#L17)
* `DevBleakScanner` uses `asyncio.create_task(self.run())`. Does this work or does it require some change? [code](https://github.com/ttu/ruuvitag-sensor/blob/8082ff4c20cc5b2110ec245d565cdd94408f1619/ruuvitag_sensor/adapters/development/dev_bleak_scanner.py#L35)
* There was the issue with `asyncio.run()` and `Bleak` adapter with Python 3.9. Can we find a solution for this or in worse case drop support for 3.9?  https://github.com/ttu/ruuvitag-sensor/issues/186#issuecomment-1271824987

terop, 2024-11-11:
Sorry for the late reply. Please find some answers to your questions below. I will open a PR to update the examples shortly.

> Bleak works normally after the change to asyncio.run().

This seems to be the case per my testing.

> RuuviTagReactice uses loop = asyncio.get_event_loop(). Should we use get_running_loop here?

I didn't see any error when trying to run an example using `RuuviTagReactive` still having `asyncio.get_event_loop()`. Per the [documentation](https://docs.python.org/3.12/library/asyncio-eventloop.html#asyncio.get_event_loop) using `get_event_loop()` is still okay when an event loop is running and this seems to be the case with `RuuviTagReactive`.

> DevBleakScanner uses asyncio.create_task(self.run()). Does this work or does it require some change?

Per Python documentation using `asyncio.create_task()` should still be okay.

> There was the issue with asyncio.run() and Bleak adapter with Python 3.9. Can we find a solution for this or in worse case drop support for 3.9?

This is a tricky question. The fix proposed by you in the issue indeed works but should it and how should it be applied to the code? Some kind of Python "#ifdef" would be possible but it seems quite dirty to me. About dropping Python 3.9 support: are there any statistics which would show many users still using Python 3.9 downloading this package?

PR #257 (2024-12-13): refactor: fix asyncio DeprecationWarnings
Calling `get_event_loop()` in Python 3.12 and newer when no event loop is running is deprecated. The recommended option is to use `asyncio.run()` instead.

Fixes #246.
closed 2024-12-13 by ttu