ttu/ruuvitag-sensor #247: Switching from hcitool to bluetoothctl

RedFoxFinn, 2024-06-04:
I am wondering if it would be possible to make this change since [this could present a problem](https://stackoverflow.com/a/71929821) with trying to scan tags with a BT 5.x device.

Not exacly sure yet would it be a small and trivial change or painful journey to implement that.

ttu, 2024-06-16:
Hi. I think this sound like a good addition 👍 

We would need to implement an own communication module for 
`bluetoothctl` in a same way as there is for `hcitool` and add a configuration option to choose the used tool. https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/adapters/nix_hci.py
https://github.com/ttu/ruuvitag-sensor/blob/dee46fbd36f01b67e22e29a81378a7b445247920/ruuvitag_sensor/adapters/__init__.py#L11

I can inspect this option more when I have time and post my findings here., but if some one finds this tasks interesting, feel free to start to work on this.

ttu, 2025-01-20:
Work in progress: https://github.com/ttu/ruuvitag-sensor/tree/feat-bluetoothctl-adapter